christianhaitian/arkos #1268: BBC Micro Retroarch core now available

SuperBadger77, 2025-02-04:
The BBC Micro core (B2) is now available. Could this system now be added to ArkOS?

christianhaitian, 2025-02-04:
Nice.  I'll review it for inclusion for the next update possibly next month.  Thanks for the heads up.

https://docs.libretro.com/library/b2/

christianhaitian, 2025-02-05:
I've built the core and made it available in the core repo.  You can test it by downloading it through the retroarch.  Make sure you update the core info files first so it will appear in the core download menu.  Let me know how the testing goes.

christianhaitian, 2025-02-07:
So currently, it only possible to start this core with a game only after a swapfile is created of at least 350MBs in size.  Yet the emulator never makes use of the swap file.  I've raised the issue with the core creator and he/they was able to replicate the issue and is investigating it further.  Also, the emulator is missing the ability to assign the * key which makes games like Castle Crasher not very playable since you can clime ladders.  I'm going to see if a simple modification of the source can resolve that.

SuperBadger77, 2025-02-07:
> So currently, it's only possible to start this core with a game only after a swapfile is created of at least 350MBs in size. Yet the emulator never makes use of the swap file. I've raised the issue with the core creator and he/they was able to replicate the issue and is investigating it further. Also, the emulator is missing the ability to assign the * key which makes games like Castle Crasher not very playable since you can't climb ladders. I'm going to see if a simple modification of the source can resolve that.

Yeah - I saw the issue about the swap file - odd one. It's a new core so there will be issues. I've been trying to promote it's use so hopefully any issues that get raised will hopefully get resolved. If you can't use the '*' key that is going to affect quite a few games which use the classic Z X / * controls. Thank you for getting involved and trying to get it to work.

In related news, I don't know if you remember I tried getting the the repo shared for the Beebem core that was held in private by the owner on RecalBox. I have used the core on RecalBox and it is really good, the only issue being the sound didn't work. I posted on the RecalBox about the new core being available and it turns out the owner Bkg2k made the repo public 4 months ago and may even look into fixing it: https://gitlab.com/recalbox/recalbox/-/issues/2052 . If this gets fixed it would be another good BBC option.

christianhaitian, 2025-02-08:
This has been added with today's update.

closed 2025-02-08 by christianhaitian

SuperBadger77, 2025-04-18:
Hi, finally got round to testing this out. Downloaded the latest image for my RG351MP, got wifi running with a dongle, ran all the updates to the latest. I changed sd card to sd2 and have some GBA and NDS games on the card which appear in ES. I've moved a load of games with the correct extension into the \bbcmicro\ folder on SD2 but the games don't appear in ES but the GBA NDS ones do. Any idea what I'm doing wrong? All games are .ssd format or .dsd.

christianhaitian, 2025-04-18:
Try switching to main sd, reboot, then switch to sd2 and see if they show up after that.

SuperBadger77, 2025-04-18:
> Try switching to main sd, reboot, then switch to sd2, then reboot again and see if they show up after that.

Boom! That worked, thanks. The old turn if off and on whilst switching rom sources. 

I wonder if I can load my RA key config files into the b2 config folder rather than having to use the ArkOS key to pad mapper for all the games. Will test it out. Thank you.