christianhaitian/arkos #1269: [Featrue] Full explain folder structure roms SD-card

poudenes, 2025-02-06:
Hi All,

I have a new R36s and a Samsung 128GB 2nd SD-card. Free memory about 2GB.
I see many folders i don't know what to do with:
- downloaded_images
- downloaded_video

There are more folders for example:
- # PT-BR #

I cant find a article where the folder structure is described and their function.
Can someone help me out or maybe add this to the Wiki?

Also i see many .MP4 files around, can you delete those, what is the function about this?

Greets,
Peter

christianhaitian, 2025-02-06:
Hello,

I can only explain what is included with the OS i curate here in this git.  What the manufacturer added when they distributed the OS with their device I can not explain.  You'll have to ask them.

None of the files and folders you listed above are familiar to me.

Thanks,

AeolusUX, 2025-02-07:
> Hi All,
>
> I have a new R36s and a Samsung 128GB 2nd SD-card. Free memory about 2GB. I see many folders i don't know what to do with:
>
> * downloaded_images
> * downloaded_video
>
> There are more folders [...]

PT-BR folder are for Portuguese and Brazilian Games.
downloaded_images and downloaded_video are both folders that stores the videos and images for use by the gamelist.xml

Christian is right the manufacturer added them when they distributed the OS with their device in this case you should ask the community about it rather than the developer here this is for bugs and issues on the operating system.