christianhaitian/arkos #1270: Cyrillic

rinart, 2025-02-06:
Device: RG353VS

christianhaitian, 2025-02-06:
See if the attached resolves this issue for you.  Restore the attached arkosbackup.tar.gz file.  The instructions for this are as follows:

1. Take the attached compressed file (arkosbackup.tar.gz) and put it in your roms/backup folder. Do not uncompress the contents! Just place the file as is in your roms/backup folder. If you don't have a backup folder in the roms folder, just create it.  If you're using a 2 SD card setup on your device, this must be done from SD1(TF1 Slot).  

2. Then go into the Emulationstation Menu, scroll to Options, hit A, then go to Advanced, hit A, then scroll all the way down to Restore Settings, hit A.

3. Once completed, restart your device and see if this issue is resolved.

Let me know your results.

[arkosbackup.tar.gz](https://github.com/user-attachments/files/18696910/arkosbackup.tar.gz)

rinart, 2025-02-06:
Nothing happened. I guess something happened with the system font. The font from the theme is displayed correctly.

rinart, 2025-02-06:
Oh no. Now RetroArch and RetroArch 32 settings overwrite each other.

christianhaitian, 2025-02-06:
The backup file I provided doesn't impact anything with retroarch or retroarch32.  It only replaced emulationstation and added a fallback.ttf files.

rinart, 2025-02-06:
[image]

rinart, 2025-02-06:
I installed the system cleanly. Some settings were made, such as changing the interface from rgui to ozone.
After resetting the settings I can no longer do this with ReatroArch 32. Mystic.

rinart, 2025-02-06:
I don't know exactly, but the reset helped. I did it again, rebooted and switched from theme to theme.
But the other problem above remains. 😂

closed 2025-02-06 by rinart