christianhaitian/arkos #1273: emergency system shutdown

Ulyapka, 2025-02-08:
After the last update, the system shuts down the device at random times during game play, as if the battery is low. If the power cable is not connected, the console cannot be switched back on. This happens especially quickly when running GameCube and Dreamcast games. I checked on other operating systems and found no similar problems. Anbernic RG353VS device, ArkOs 2.0 version (02022025)

christianhaitian, 2025-02-08:
That's a battery issue.  Nothing in the OS will trigger a shutdown due to low battery.  At most, you'll get a warning.

closed 2025-02-08 by christianhaitian

Ulyapka, 2025-02-08:
There are no such problems in a stock system. Also in ArkOs this problem did not occur until the last update. There is nothing wrong with the battery, the problem is directly in the latest update.

christianhaitian, 2025-02-08:
Is the unit going through a shutdown procedure when this emergency system shutdown occurs?  Or does it seem like power just gets cutoff?

Ulyapka, 2025-02-09:
It's like the power just goes out

christianhaitian, 2025-02-09:
That's a battery issue.  No software update that was done would cause an issue like this especially since you stated, it only happens when the unit is not connected to power.

Ulyapka, 2025-02-09:
It's very strange. Why then was this not there before the update and why is this problem not there when playing on the stock OS?

christianhaitian, 2025-02-09:
It could have been there but just didn't present itself until recently.  Not far fetched as i'm on my 3rd battery with my rg353v unit.  I can't explain why it's not with the stock OS.  Perhaps you didn't play it long enough or with a demanding emulation like Dreamcast or Gamecube to exhibit the issue?

Ulyapka, 2025-02-09:
I had an idea that it was because the GameCube is demanding, but when the console shut down while playing GameBoy Advance, I decided to write to you about it. For a test I played Shenmue 2 on the Dreamcast with the stock OS for about 20 minutes and nothing happened, but in ArkOS the shutdown happens as early as the opening cutscene. I want to try to install an earlier version of ArkOS, but I'm afraid of losing my game saves, as the saves made when playing on ArkOS are not available on the stock system for some reason. If everything will work fine on the earlier version of ArkOS, then the version with the bad battery will be wrong.