christianhaitian/arkos #1274: Keyboard font

lauren7ino, 2025-02-10:
Strange issue, I just updated my rg353m and the rgb30, and on the rgb30, the keyboard is missing special characters like Return, etc.

[image]

christianhaitian, 2025-02-10:
Not seeing this issue on my rgb30 updated to the 02092025 update.

[image]

lauren7ino, 2025-02-10:
Is the font related to the current theme?

christianhaitian, 2025-02-10:
I don't think so but I'm not 100 percent sure.

lauren7ino, 2025-02-11:
1)
It seems that the rg353m updated to the latest version, but the rgb30 (that was updated simultaneously) didn't bring ArkOS to the latest version (is this normal?)

The keyboard (encoding?) bug was related to a previous version of ArkOS.

2)
Yes, it seems that the keyboard uses the same font of the ES menus.

Problem solved, sorry to bother you.

christianhaitian, 2025-02-11:
Don't know why it didn't update to the latest version.  Usually that's because of a download issue.  Perhaps github lost connectivity during the process?  Who knows.  Happy it's solved now.  Enjoy.

closed 2025-02-11 by christianhaitian