christianhaitian/arkos #1275: Font corruption on ES in cyrilic translate (russian and ukrainian) on ArkOS 2.0 (02022025)

VolodymyrDA, 2025-02-10:
Font corruption on ES in cyrilic translate (russian and ukrainian) on ArkOS 2.0 (02022025). All was good in previous version 24.12.2024.
Also can you add this translation to system?

[image]

VolodymyrDA, 2025-02-10:
[translate.zip](https://github.com/user-attachments/files/18741437/translate.zip)

christianhaitian, 2025-02-10:
> Font corruption on ES in cyrilic translate (russian and ukrainian) on ArkOS 2.0 (02022025). All was good in previous version 24.12.2024. Also can you add this translation to system?
> 
> [image]

You're running the aeoulusUX community image on the r36s.  You need to report this issue to that repo here: https://github.com/AeolusUX/ArkOS-R3XS/issues

christianhaitian, 2025-02-10:
> [translate.zip](https://github.com/user-attachments/files/18741437/translate.zip)

I've added these translations to the emulationstation git.  It will be included in the next update.  Thank you

closed 2025-02-10 by christianhaitian