christianhaitian/arkos #1276: Feature - web browser

dani7959, 2025-02-10:
I got quite excited when I discovered the NetSurf browser port. A browser can make a huge difference in a device.
But when actually trying to use NetSurf I felt a bit limited:
- cannot change homepage (not even in the Choices config file)
- cannot bookmark a page
- controls are working but still very basic
- cannot use google (complains about unusual traffic, might be specific to my config)

Looking into non-javascript browsers, Links ( [http://links.twibright.com/](http://links.twibright.com/) ) seems to be a very popular choice.
It's been around since 1999 and is still updated as of July 2024.
It's optimised for keyboard shortcuts so it will make a very good fit for controllers.
Tested side by side with NetSurf in a desktop environment looks to be the better option.
There are pretty good installation instructions under [http://links.twibright.com/download.php](http://links.twibright.com/download.php)
It requires very few packages, most of which are already available in Arkos, so there it can be a native program.

I thought I could give it a try and install Links natively, and maybe you could give some insight.
Maybe you tried this already, or maybe you could have a look at the required libraries and advise if they can work or not under Arkos.

christianhaitian, 2025-02-10:
I am not interested in adding any internet browser to the distro.

closed 2025-02-10 by christianhaitian

lauren7ino, 2025-02-11:
Just saying, a great place to add a web browser would be as a Port (PortMaster)