christianhaitian/arkos #1277: Botting with 0% battery

lauren7ino, 2025-02-12:
Hi,
Is there a way ES (of even before ES loada) you could visually alert the user that the system is out of battery?

Yesterday I booted my RGB30 with no battery left.
I believe the sleep mode depleted the battery overnight.
I was updating RetroArch and 2 seconds before crashing I saw 0% on the RA screen.
Consequence: SDCard corrupted and unbootable.

Thanks, keep up the good work

christianhaitian, 2025-02-12:
There is no visual indicator prior to loading ES, however, when you load into retroarch, you should've gotten a verbal low battery warning if you left that feature on in the Start Menu>Advanced section.  Also, did you not enable deep sleep in the Options>Advanced section or is your unit not stable with deep sleep?

lauren7ino, 2025-02-12:
Only when I get back from work I can confirm.
...
I usually have the volume at 0-1% at home until I actually play a game (sounds attract kids), that's why I asked for a visual warning.
Every other system I know doesn't allow for a full boot when the battery is too low.
Maybe ES could auto-shutdown when detects it's below x% to prevent damaging the card?
Thanks

christianhaitian, 2025-02-12:
Since the battery reading is not accurate and can fluctuate pretty widely, that would probably cause other issues.  I've not run into corrupted installs when using good quality SD cards like Sandisk or Samsung.  Are you sure your unit is not booting because the battery is still not charged enough?

lauren7ino, 2025-02-12:
Unfortunately I destroyed all evidence, because I already saved to that SanDisk an IMG backup I had made in December.
I can't even tell if it was deep sleep enabled (probably not)

ES is probably not where the battery fluctuates most.
I believe it would be safe to set it to 5%, because as the user launches an emulator probably will drop to zero in seconds.

Anyway, thanks. I love ArkOS, I won't even consider buying devices that don't support ArkOS 😊

closed 2025-02-12 by christianhaitian