christianhaitian/arkos #1278: Battery dies at random after one of last updates.

j0ngraham84, 2025-02-12:
Around the end of January I installed an update that started some battery problems. It just dies at random . It will be at 80 percent and it will just shut off. Usually it happens when exiting a game but it has happened in game too. I read in the change log about some reverted battery issues but I wasn't having any problems with it not charging past 50 percent. I installed the latest updates yesterday and it just happened again. I have to plug into a power source and the light will come back on and then it will work for about 10 to 20 min and then randomly die. I tried unplugging for the battery for 10 sec and that didn't help either. My guess is that maybe it's related to those reverted battery changes the change log mentioned?

j0ngraham84, 2025-02-12:
Also it's the 353m

christianhaitian, 2025-02-12:
If you updated to the latest 02092025 update, those battery changes have been reverted.  Make sure you update to that and unplug the battery for 10 seconds then fully charge the battery to 100 percent.

j0ngraham84, 2025-02-12:
I did install that update. I removed the battery for about a minute and reconnected it. I did not charge it to 100 percent before I retested it. Maybe that's it. I'll try that next. And just in case it matters I went from the 1/31/25 update directly to the 2/9/25 update. Im assuming that included the updates from 2/1 ,2/2 and 2/8 as part of 2/9 . If that doesn't work I'll try replacing the battery next. I think I remember hearing that the 350m battery is compatible with the 353m so if that's true I will probably try that too. Thanks for the quick reply.

christianhaitian, 2025-02-12:
Updated can't be skipped.  So if you're on 02092025, you have all the previous updates.  I believe the 351p and 351m battery is the same.  They all have the anbernic 3500mah battery.  I believe it's the same for the 351mp unit as well.  The 353v has a smaller battery.

j0ngraham84, 2025-02-14:
So it's looking like it's the original battery. I again removed it this time for about 10 min, reattached and charged it to 100 percent. And once again it died within 10 min while playing a game of hoops96 on final burn. It will not power back on until you plug it into a charger and it shows 98% power. I just replaced the battery with the one from the 351m . Im going to charge it to 100 % percent before I test it again. I'll update with the results. I will say the 351m looks like it had a higher quality battery. It had more insulation on the wires and it seemed more solid than the 353 battery which felt very bendable as I gently removed it from the adhesive on the backplate.

j0ngraham84, 2025-02-15:
I installed the 351m battery. I even flashed a new sd card with the 2//9 image . I thought I had fixed. It finally lasted more than 15 min without dying . I played PlayStation for an hour or so and when I exited back to emulation station it just dies again. I think it's hardware glitch. All I can do now is try and replicate on the stock os.

christianhaitian, 2025-02-16:
Hmm, very odd.

christianhaitian, 2025-02-16:
See if replacing the rk3566-OC.dtb file in the BOOT partition with the one in the attached zip file resolves the issue.  The only difference is that it disabled the native rumble support added with the 01312025 update.

[rk3566-OC.zip](https://github.com/user-attachments/files/18813202/rk3566-OC.zip)

spirit-in-exile, 2025-02-16:
Encountered a similar issue when performing a fresh flash of 02//07/2025 on my RG353M yesterday: Device would randomly shut off during various tasks, despite battery indicator readings above 50%. Different cards = same results. Battery reset + full-charge later = things seem to have stabilized, no further unexpected losses of power. 

Oddly, I also updated my wife's RG353P (I know: Unsupported) via OTA update, and after some testing, it's exhibited no ill effects.

So for me, this problem only seemed to occur on the freshly flashed install, and so far seems to have been resolved with the battery reset and full-charge guidance given [here](https://github.com/christianhaitian/arkos/issues/1278#issuecomment-2654653121) and in the 02/01/2025-2 [Changelog](https://raw.githubusercontent.com/christianhaitian/arkos/main/changelogs/rg353m-changelog) entry.

Will update this commend if that changes.

j0ngraham84, 2025-02-16:
I also noticed that the last time it died it failed to update my memory card on PlayStation. I had just finished a good hour or so gaming session. I saved and exited to es and then it died .. Anyway I replaced the .dtb file . So far it has not died since . Ive exited out a few games and it's remained on. I even let it idle for a hour and it's  good. I played the same game that crashed it last time for about 20 min , saved a few times, exited back to es a few times and so far it's not crashed. I plan to give it some heavy usage later this evening so if anything happens I'll update for now it's working good again after replacing that .dtb file

christianhaitian, 2025-02-16:
> I also noticed that the last time it died it failed to update my memory card on PlayStation. I had just finished a good hour or so gaming session. I saved and exited to es and then it died .. Anyway [...]

Ok.  Thanks for the update.  I found one other change that was in the source code unrelated to vibration that may have also attributed to this issue.  It was a change to address issues with some units not being able to successfully boot ArkOS that this person here was able to address in this issue [here](https://github.com/christianhaitian/arkos/issues/837#issuecomment-2240323719).  If you can, please use the dtb attached and let me know if this causes your unit to shut off unexpectedly or not.

For RG353M units, use this one-->[rk3566-OC-353m.zip](https://github.com/user-attachments/files/18816274/rk3566-OC-353m.zip)
For RG353V units: use this one--->[rk3566-OC-353v.zip](https://github.com/user-attachments/files/18816275/rk3566-OC-353v.zip)

HVS2, 2025-02-17:
Running into the same problem on my RG353V. 
This started 2-3 updates ago for me. I fire up a game and, 10min or so into it, the whole device dies completely. Then won't power up or reset again.

Quick connect with a charger gets the LEDs to turn on and only then allows the device to turn on again. 
Tried a new SD card with a fresh ArkOS install, same problem.

Opened the device earlier and unplugged battery, however didn't see the comment to charge it to 100% right after. Maybe I'll try that later. 

Another thing I also noticed is an insanely high battery drain now even idling in ES. Can practically watch percentage drop.
Under Android and on stock OS everything plays without problems and battery life is as it's always been.

------
Just tried another SD card with a fresh image, replaced the .dtb with the one from the above post before the first boot.
ArkOS now shows the correct percentage (same as stockOS).

Some quick testing (my goto is SNES Kirby with its demos on loop):
Looking good so far
Battery now drains correctly (about 2% for 15-20ish min of game time)
Haven't had a crash yet in about 30min of uptime. Will test a bit more tomorrow.

christianhaitian, 2025-02-17:
> Running into the same problem on my RG353V. This started 2-3 updates ago for me. I fire up a game and, 10min or so into it, the whole device dies completely. Then won't power up or reset again. [...]

So you’re using the dtb i provided [here](https://github.com/christianhaitian/arkos/issues/1278#issuecomment-2661616417)?

Ulyapka, 2025-02-17:
That's exactly my case. Replacing the file rk3566-OC.dtb solved the problem. Tested the same way as before on Dreamcast emulator. Previously the power was cut off in the opening cutscene of Shenmue 2, now it successfully works further and you can play. Thanks for solving the problem!

AlgoAlex-K, 2025-02-17:
> > Running into the same problem on my RG353V. This started 2-3 updates ago for me. I fire up a game and, 10min or so into it, the whole device dies completely. Then won't power up or reset again.
> > Quick connect with a charger gets the LEDs to turn on and only then allows the device to turn on again. Tried a new SD card with a fresh ArkOS install, same problem.
> > Opened the device earlier and unplugged battery, however didn't see the comment to charge it to 100% right after. Maybe I'll try that later.
> > Another thing I also noticed is an insanely high battery drain now even idling in ES. Can practically watch percentage drop. Under Android and on stock OS everything plays without problems and battery life is as it's always been.
> > Just tried another SD card with a fresh image, replaced the .dtb with the one from the above post before the first boot. ArkOS now shows the correct percentage (same as stockOS).
> > Some quick testing (my goto is SNES Kirby with its demos on loop): Looking good so far Battery now drains correctly (about 2% for 15-20ish min of game time) Haven't had a crash yet in about 30min of uptime. Will test a bit more tomorrow.
> 
> So you’re using the rg353v dtb i provided [here](https://github.com/christianhaitian/arkos/issues/1278#issuecomment-2661616417)?

Correct! That exact file

j0ngraham84, 2025-02-18:
I tried the second .dtb on the SD card I was using when I first started having problems and now it's running fine. I've been playing PlayStation off and on for a few hours it's been behaving fine. I also never had any problems with the first .dtb file posted that I used on the new SD card . Both seem to fix the issue on different cards.

christianhaitian, 2025-02-18:
Ok.  Thanks for confirming.  Interestingly, none of my 353 devices have issues with any of these different dtbs.  It handles all of them fine. Very odd.

Asuza, 2025-06-03:
Should I let the battery drain fully before disconnecting it for at least 10 seconds?. I'm having major battery drain issues as well on RG353M. I just updated to the latest ArkOS and it's still an issue. Thanks!

christianhaitian, 2025-06-03:
If you were on a mainline distro before using ArkOS and never disconnected the battery, just disconnect the battery for 10 seconds.  No battery drain needed.  If that doesn't fix the issue, try this file here: https://github.com/christianhaitian/arkos/issues/1278#issuecomment-2661616417

j0ngraham84, 2025-06-04:
It's been a few months now and just wanted to update that I haven't had any more battery problems. I was worried if installing updates would revert it but it did not. It's been smooth sailing since I replaced that .dtb file back in February.

Wysie, 2025-07-05:
Just writing to update that I did a fresh install of ArkOS on my RG353V and updated it, and faced this issue (device would randomly shut off after a while) until I replaced the dtb file with the file mentioned [here](https://github.com/christianhaitian/arkos/issues/1278#issuecomment-2661616417). It's been running fine for about a day since then. Thanks!

amansn7, 2025-10-25:
> > I also noticed that the last time it died it failed to update my memory card on PlayStation. I had just finished a good hour or so gaming session. I saved and exited to es and then it died .. Anyway I replaced the .dtb file . So far it has not died since . Ive exited out a few games and it's remained on. I even let it idle for a hour and it's good. I played the same game that crashed it last time for about 20 min , saved a few times, exited back to es a few times and so far it's not crashed. I plan to give it some heavy usage later this evening so if anything happens I'll update for now it's working good again after replacing that .dtb file
> 
> Ok. Thanks for the update. I found one other change that was in the source code unrelated to vibration that may have also attributed to this issue. It was a change to address issues with some units not being able to successfully boot ArkOS that this person here was able to address in this issue [here](https://github.com/christianhaitian/arkos/issues/837#issuecomment-2240323719). If you can, see if replacing the rk3566-OC.dtb file in the BOOT partition with the one in the attached zip file for your device causes your unit to shut off unexpectedly or not.
> 
> For RG353M units, use this one-->[rk3566-OC-353m.zip](https://github.com/user-attachments/files/18816274/rk3566-OC-353m.zip) For RG353V/VS units: use this one--->[rk3566-OC-353v.zip](https://github.com/user-attachments/files/18816275/rk3566-OC-353v.zip)

Does this also solve the issue here :  https://github.com/christianhaitian/arkos/issues/1020

amansn7, 2025-10-25:
> > I had this exact issue and it was fixed upon replacing the dtb file with the one here: [#1278 (comment)](https://github.com/christianhaitian/arkos/issues/1278#issuecomment-2661616417)
> > However, I am now getting duplicated images at the bottom of my RG353V, as described in this issue thread: [#1020](https://github.com/christianhaitian/arkos/issues/1020)
> > When I replace my dtb using the instructions in the thread above, the duplicate image goes away but the power issues return.
> > Is there a dtb file out there that fixes BOTH the duplicate images at the bottom during extended gameplay AND the power issues? It seems as though I have to pick one or the other.
> > Thanks!
> 
> [rk3566-OC.zip](https://github.com/user-attachments/files/23140483/rk3566-OC.zip)
> 
> I've tried combining both the dtb files. I am currently running the device 20-30 minutes in, no issues so far. I have no idea about how these things actually work, i used chatgpt to do some digging around both files.
> 
> Would love to get confirmed by the developers here.

[rk3566-OC-353v.zip](https://github.com/user-attachments/files/23140572/rk3566-OC-353v.zip)

If this is helpful at all!

Manwe-fv, 2025-10-25:
> [rk3566-OC.zip](https://github.com/user-attachments/files/23140483/rk3566-OC.zip)
>
> I've tried combining both the dtb files. I am currently running the device 20-30 minutes in, no issues so far. I [...]

Awesome - I will try this out. Appreciate it!

Haddouken, 2025-11-13:
I play with Achivements, so i moved on this week on my 353vs Stock OS (outdated Retro Arch (wich is a demand for cheevos atm)) to Ark OS, and now im having this issue as well. So far i have used the DTBs from [amansn7](https://github.com/amansn7) .
The first one gave me 'ghosting' on text (stock ark dtb did as well, only -bright fixed it) and the second no ghosting but shutted down today again at 30% battery after pressing 'update lists'. Which one should i try next? [rk3566-OC-353v.zip](https://github.com/user-attachments/files/18816275/rk3566-OC-353v.zip) or [rk3566-OC.zip](https://github.com/user-attachments/files/18813202/rk3566-OC.zip) ?