christianhaitian/arkos #1281: PortMaster not starting on a fresh install of ArkOS 02092025

dani7959, 2025-02-13:
I noticed that fresh images of ArkOS were published, so I did a fresh install on a RG353VS.
Setup ES and RA to my liking. Changed roms to SD2.

Then went on and started PortMaster, but it would not go past the splash screen.
The actual ports are starting OK.

PortMaster log below:

Failed to create secure directory (/run/user/1002//pulse): No such file or directory
2025-02-13 23:00:32.368 | ERROR    | __main__:<module>:2078 - An error has been caught in function '<module>', process 'MainProcess' (1210), thread 'MainThread' (548670066704):
Traceback (most recent call last):

> File "./pugwash", line 2078, in <module>
    exit(main(sys.argv))
         â”‚    â”‚   â”” ['./pugwash']
         â”‚    â”” <module 'sys' (built-in)>
         â”” <function main at 0x7fbc39f3b0>
  File "./pugwash", line 2039, in main
    pm = PortMasterGUI()
         â”” <class '__main__.PortMasterGUI'>
  File "./pugwash", line 624, in __init__
    if not is_process_running(f"love.{capabilities['primary_arch']}"):
           â”” <function is_process_running at 0x7fbc3f6cb0>

KeyError: 'primary_arch'
0: 1
{'name': 'ArkOS', 'version': '02092025', 'device': 'rg353v', 'resolution': (640, 480), 'analogsticks': 2, 'cpu': 'rk3566', 'capabilities': ['power', '4:3', '640x480', '1gb', 'en_US']}
arm_release_ver of this libmali is 'g2p0-01eac0', rk_so_ver is '3'.arm_release_ver of this libmali is 'g2p0-01eac0', rk_so_ver is '3'.arm_release_ver of this libmali is 'g2p0-01eac0', rk_so_ver is '3'.arm_release_ver of this libmali is 'g2p0-01eac0', rk_so_ver is '3'.Attempting to delete file at path: /opt/system/Tools/PortMaster/utils/pmsplash/stopsplash
File 'stopsplash' deleted successfully.

christianhaitian, 2025-02-13:
I can't replicate the issue on my end using a fresh install.  Not sure what happened on your end but  a fresh install of PortMaster should resolve it.

closed 2025-02-13 by christianhaitian

dani7959, 2025-02-14:
Yes, a fresh install of PortMaster solved it. Good to know it's just on my end. Thanks!