christianhaitian/arkos #1282: Hypseus-singe does not accept commands for American Laser Games (and fix for it)

GitGud-art, 2025-02-14:
Hypseus-singe is working properly with daphne games inside `/rom/daphne/` folder but does not accept handheld/gamepad inputs in singe games, which are placed in `/rom/alg/`.

This can be easily circumvented by the user by adding a commands file to the singe games, but the arguments are not being passed to the singe.sh script in `/usr/local/bin/singe.sh`.

The best way to fix it can be by going about it in a few ways: 
* to add gamepad support as default to ArkOS: 
     * add `-gamepad \` option to LD_PRELOAD var;
* to add support for reading the custom singe game commands:  
     * add `$(cat "$HYPSEUS_SHARE/$SINGEGAME/$SINGGAME.commands") \` 
* to add both gamepad as default and support for reading the custom singe game commands:  
     * add `-gamepad $(cat "$HYPSEUS_SHARE/$SINGEGAME/$SINGGAME.commands") \` 

This was tested in an r36s clone with a custom ArkOS build, but this issue also pertains the main repo.

christianhaitian, 2025-02-14:
What game did you encounter this issue with?  input works fine with ninja hayate and time gal on my end.

GitGud-art, 2025-02-15:
All tested games, 10 in total. Besides, the `singe.sh` script does not pass arguments to the commandline execution. `daphne.sh` does and uses a different way to pass the args by defining a variable with the reads. I guess it was not copied over to `singe.sh` previously for some good reason. That solution can also be copied among scripts.

By the way, I made a typo on `$SINGGAME.commands` . Should read `$SINGEGAME.commands`.

christianhaitian, 2025-02-15:
- gamepad and reading $SINGEGAME.commands is already included in the singe.sh file I provide with mainstream ArkOS.  See here: https://github.com/christianhaitian/rk3326_core_builds/blob/rk3326/shell-scripts/singe.sh

GitGud-art, 2025-02-15:
I see. Thank you for looking into this. I was assuming the r36s port was using the full build of this arkOS. My bad. I will link in the proper project.

closed 2025-02-15 by christianhaitian