christianhaitian/arkos #1285: Ukrainian (uk-UA) and Russian (ru-RU) locales performance 90~95% from using ArkOS.

SerjGeek, 2025-02-18:
Using ArkOS as a base, reworked these two locales manually by calculating and using almost all available strings and grammar, individually on devices for each locale at the same time. Minor adjustments are possible in the future, but the main thing is ready.

[local.zip](https://github.com/user-attachments/files/18855255/local.zip)

[image]

VolodymyrDA, 2025-02-20:
Can you describe more details, becouse i adapted this translation update to ES

SerjGeek, 2025-02-20:
Версія(ревізія) є і наприклад в Анберніках це іноді має значення бо бувають різні завантажувачі системи під різні екрани,  Якщо в тебе на R35S/R36S то там "портована" ArkOS, ти скинув локалізацію зі свого R35/36S де тієї інформації може і не бути як можливо і інших пунктів... А загалом скинув там +-50% переклад, я робив це майже з нуля впродовж двох тижнів, так що видаляй своє, а моє має своє місце тут в оригінальній ArkOS системі для Anbernic. Так я за ради цікавості ще придбав R36S, так що далі буде😄Ну а відсутність переводу в інших пунктах то справа часу, все одразу охопити неможливо.

/

There is a version (revision) and, for example, in Anberniki, it sometimes matters because there are different system loaders for different screens. If you have ArkOS "ported" on R35S/R36S, you reset the localization from your R35/36S, where that information may not be there as well as other items... And in general, I reset +-50% translation there, I did it almost from scratch for two weeks, so delete yours, and  mine has its place here in the original ArkOS system for Anbernic.  So, for the sake of curiosity, I also bought R36S, so what will happen next😄 Well, the lack of translation in other points is a matter of time, it is impossible to cover everything at once.

[image]

VolodymyrDA, 2025-02-21:
@SerjGeek  
Друже, я тобі раджу для початку ознайомитись що таке GIT , і як зним працювати . Це дуже допоможе тобі в твоїй власній розробці і з інтеграціею ії з іншими.
Основне її призначення вирішення конфлікту версій.
Для створення фікса ти повинен відштовхуватись від поточної версії перекладу, а мій переклад адаптований і перевірений на r36s . Якщо в інших системах відсутні якісь пункти добав їх, не порушуючи поточний. 
А ти пробуєш викласти іншу версію з косяками які я виправив.
Це щодо технічної частини.
Щодо якості перекладу. 
1. Роби переклад відштовхуючись від оригіналу. Не треба "обране" називати "топ" тому що терміни повинні бути сталими, вони можуть вживатись в іншому контексті, наприклад "Обране завжди зверху", і також добавлятись по ходу розробки.
Без крайньої потреби робити цього не варто.
На рахунок  твого прикладу з екраном тото неварто роздувати назву,  "параметри екрану" досить точно характеризують що всередині.
Також твій переклад некоректний, не існує терміну "годинний пояс" , насправді він називається "часовий пояс". І таких ляпів в тебе вистачає.  Також дословного  перекладу недостатньо, треба його робити враховуючи технічну частину, а то буде "заховати сховані системи"
Так що використовуй існуючий переклад, можеш навіть зробити pull request в гілку ES щоб було все по феншую. 

My friend, I advise you to start by learning what GIT is and how to work with it. This will greatly help you in your own development and in integrating it with others.
Its main purpose is to resolve version conflicts.
To create a commit, you should base it on the current version of the translation, and my translation is adapted and tested on r36s. If some items are missing in other systems, add them without breaking the current structure.
But you're trying to push a different version with mistakes that I have already fixed.
Regarding the technical part:
    Do the translation based on the original text. There is no need to call "Favorites" "Top" because terms should remain consistent. They might be used in different contexts, such as "Favorites always on top," and could also be added during development.
    Unless absolutely necessary, avoid changing established terms.
    Regarding your example with the screen settings—there's no need to overcomplicate the name. "Screen parameters" accurately describes what’s inside.
    Your translation is also incorrect. The term "годинний пояс" (hourly zone) doesn’t exist—it should be "часовий пояс" (time zone). And you have plenty of similar mistakes.
    A literal translation is not enough; it should be done with the technical context in mind. Otherwise, you end up with things like "hide hidden systems."
So, use the existing translation. You can even submit a pull request to the ES branch to keep everything properly organized.

SerjGeek, 2025-02-21:
Взагалі не розумію про що ти, в тебе самого переклад невірний, від чого ти там відштовхуєшся я тебе взагалі не питав, сам причипився...) Теж навчись для початку, лише наклеп завів тут зі своїм r36s котрий і поряд з анберніком не стоїть як виявилося, щей свій кривий переклад втикнув, може заспокоїшся і будеш свій переклад покращувати на тому r36s? бо я його на ньому тел побачив) Почни з UK UA для початку виправ це, потім я з тобою продовжу спілкування, а так не бачу сенсу) А Git э Git...

SerjGeek, 2025-02-21:
Так що все буде добре, дякую за напуття, я це і так спочатку розумів ти просто не зрозумів мого посилу, я так само робив переклад на основі того, що б в більшості тем вмістився текст кнопок без скорочень наприклад те ж саме обране в обр... Хочеться мінімум скорочень, але використовуючи слова з тим самим смислом. І посил твій без того розумію, ти в своєму перекладі взагалі видалив ті слова котрі типу заважають,  опції та обирати то теж не гуд виходячи з оригіналу...

christianhaitian, 2025-03-29:
This has been included with today's update.  Thank you for the submission.

closed 2025-03-29 by christianhaitian