christianhaitian/arkos #1288: Would it be okay if I implement Netplay?

bulzipke, 2025-02-19:
Hi, I want to add Netplay to ArkOS.

I plan to create a Netplay menu and add options for enabling Netplay and configuring a relay server.

When Netplay is enabled, I want the game to launch in host mode and be exposed as an 'ArkOS Player' on lobby.libretro.com.

This should make it easier to play with others. 

What do you think?
A similar PR is as follows: https://github.com/knulli-cfw/batocera-emulationstation/pull/28

christianhaitian, 2025-02-19:
I'm not really into adding to many extra menus to ES as I'm really not a fan of that.
Why not just expand the existing Local NetPlay configuration that's available when you hold the B button down after launching a netplay compatible core from ES?

bulzipke, 2025-02-19:
I understand the direction of ArkOS. Certainly, settings like relay servers or starting as a host can also be configured within the RetroArch menu. However, personally, I find that having to configure additional settings or start in host mode within RetroArch feels like navigating a frontend inside another frontend (ES), which can be quite complex.

Additionally, many casual users may not even be aware of the 'hold B button' option. I strongly agree that a user-friendly interface should minimize user interaction as much as possible, keeping things simple and intuitive.

[image]
For example, if a device is not connected to WiFi, considered automatically creating a local AP (hotspot) so that nearby devices could connect to it and play via local network without complex configurations.  If to implement this, I was considering either calling it through crontab or implementing it as a service.

[image]
When launching a game, I considered checking it within ArkOS's `FileData.cpp`

If that fails, an ICE implementation could be used to attempt local IP connections, hole punching for P2P, or falling back to a relay server.

However, at this moment, I believe the simplest implementation would be to allow users to configure a relay server and have their game sessions listed on lobby.retroarch.com when launching a game. That's why I wanted to work on this first.

I just wanted to propose this idea, and I have no intention of interfering with ArkOS project or its direction.
ArkOS is already an amazing platform, and many players, including myself, appreciate your work!

christianhaitian, 2025-02-19:
Users would be aware of the B option if they refer to the extensive FAQ on the wiki for their device.  I appreciate the thought and detail you've put into the feature request, you most certainly can fork the ES used here and implement that on your own setup but it won't be something I'll include or provide support for moving forward.  Take care.