christianhaitian/arkos #1292: Feature Request -  Makecode Arcade ELF  file Support.

whmzsu, 2025-02-24:
Is it possible to  run  " Makecode Arcade "     ELF  file    on ArkOS  ?

The  "https://github.com/Vegz78/McAirpos/"  Project  describe "[MakeCode Arcade](https://arcade.makecode.com/?nolocalhub=1&hw=rpi&compile=rawELF) games in [RetroPie](https://retropie.org.uk/), [Recalbox 8/7](https://www.recalbox.com/) and [Batocera](https://batocera.org/), running natively as ELF executables on [Raspberry Pi OS](https://www.raspberrypi.com/software/)/Linux ARM with gamepad and keyboard support for up to 2 players."

I  tried  on the arkos ,  not work  on the "rgb 30"  device with arkos 202305 version, will hang when launch from emulation,  or with dual picture(frontend  and game ) overlay when  launch from ssh remote .
Segment fault with the arkos 202412 version.

whmzsu, 2025-02-25:
> Where did you find the executable for this?

https://arcade.makecode.com/?nolocalhost=1&compile=rawELF&hw=rpi#editor
or
https://arcade.makecode.com/beta/?nolocalhost=1&compile=rawELF&hw=rpi#editor

click download 

or any project url  with  "compile=rawELF&hw=rpi#editor"

whmzsu, 2025-02-25:
https://vegz78.github.io/McAirpos/index2

Here has some introductions.

christianhaitian, 2025-03-01:
I had time to try this today and it looks like it's hardcoded to work on Raspberry Pi units only.  It's looking for resources not available on this platform.

closed 2025-03-01 by christianhaitian

whmzsu, 2025-03-03:
> I had time to try this today and it looks like it's hardcoded to work on Raspberry Pi units only. It's looking for resources not available on this platform.

https://private-user-images.githubusercontent.com/4927222/353654810-6791b1f4-558e-440d-8ff2-408d9c47de50.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDA5NzM4MDYsIm5iZiI6MTc0MDk3MzUwNiwicGF0aCI6Ii80OTI3MjIyLzM1MzY1NDgxMC02NzkxYjFmNC01NThlLTQ0MGQtOGZmMi00MDhkOWM0N2RlNTAuanBnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDMwMyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTAzMDNUMDM0NTA2WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MjA1OGQwYmU5NjU1YTZjNjUzMjgyNmZkMmI1YjVhOGVmNWYxNTdkYWE3N2FhZDNkZGUzNzYwZTY2OTk4YjBiMSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.liQ_dKS7rOJdNeS4oLHN1toPtR7fc1XjprHJyp-LJh0

As test on older version arkos,  the elf can  launced from ssh remote ,  but  with display  overlay  (emulationstation frontend), the control is also worked.