christianhaitian/arkos #1293: Can't boot ArkOS on Anbernic RG353M

Saeniv, 2025-02-26:
Hello,

I read a lot through other issues and the wiki and reddit, but I can't solve my issue. My RG353M has no sound in the android mode, so i upgraded it to  the Firmware to v1.20, but did not solve my problem, so i wanted to install ArkOS. One time the sound worked on a youtube video, but only once. The headphone jack kind of works with a lot of noise. 

I flashed the RG353M ArkOS and the handheld won't boot, the screen stays black and the green led is shining, nothing else happens. I tried 3 different SD Cards, I tried Rufus and Win32 Image Tool, I tried different OSes like UnofficalOS, ROCKNIX and the official Linux version from Anbernic. 
With none of my attempts, I were able to boot, always the same picture, black screen green light. I fully charged the console and also left the power cord attached. 

Then I found this https://github.com/christianhaitian/arkos/issues/837#issuecomment-2241738969 and tried all of those fixes but did not work.

Does somebody else have any idea or is it just a bad unit.

christianhaitian, 2025-02-26:
Unless something is wrong with your card reader/writer, it sounds like you have a faulty unit if non of the other OS options are working for you.

ilove3gp, 2025-03-01:
> Unless something is wrong with your card reader/writer, it sounds like you have a faulty unit if non of the other OS options are working for you.

The issue is like RA cannnot be launched (black screen)
Thank you for the quick reply. I’ll try flash again with ArkOS.

Saeniv, 2025-03-01:
Thank you for your replies.

I opened the thing and measured the Power Management Chips. It seems that it has voltage stability issues during the boot of linux, but during android it's completely stable. I tried supplying via the laboratory power supply and some other things, but it did not change a thing.

christianhaitian, 2025-03-04:
Exchange it or get another one.  Not much else you can do.

closed 2025-03-04 by christianhaitian