christianhaitian/arkos #1294: RG353V gets hot really quickly after the last update

tisisandnemesis, 2025-02-26:
Hi Christian. Thanks for this awesome piece of software. I got a quite serious issue lately. After the latest update (feb one), my rg353v gets hot really quickly. When it does, it slows down a lot in all it's apps. Actually I just shut it down in fear that running it might cause damage. Thanks!

christianhaitian, 2025-02-26:
What are you doing with it when you notice it heats up?

tisisandnemesis, 2025-02-26:
> What are you doing with it when you notice it heats up?

When I notice that the front left part of the screen feels too hot, I shut it down.

christianhaitian, 2025-02-26:
We're you playing a game?  If so what game on what emulator?

tisisandnemesis, 2025-02-26:
> We're you playing a game? If so what game on what emulator?

Its when running emulators. The more taxing the emulator is, the quicker it gets hot. I first found it out when playing the ps1. It got hot, then switched to Amiga while it was still hot and it was going visibly slower.

christianhaitian, 2025-02-27:
Try this dtb file from here for the RG353V and see if that makes a difference for the temperature of your unit.

https://github.com/christianhaitian/arkos/issues/1278#issuecomment-2661616417

tisisandnemesis, 2025-02-27:
> Try this dtb file from here for the RG353V and see if that makes a difference for the temperature of your unit.
> 
> [#1278 (comment)](https://github.com/christianhaitian/arkos/issues/1278#issuecomment-2661616417)

I replaced the dtb file and let it run idle on both PS1 and Amiga emus for over 1hr. Thermally, it seems to be in a much better condition than previously. Later on, I will do some playtesting too and report back. Thanks.

theflynn49, 2025-03-03:
Hello. FYI Same effect for me on my RG353M , playing PICO8 or ARDUBOY : it gets hot rapidly and shuts down abruptly after about 10 minutes when too hot. A 5mn session with PICO8 used 5% battery, 

Replacing the .dtb file [rk3566-OC-353m](https://github.com/user-attachments/files/18816274/rk3566-OC-353m.zip)  fixed the problem. The 5mn PICO8 session now uses 0% (didn't change)

tisisandnemesis, 2025-03-04:
> Hello. FYI Same effect for me on my RG353M , playing PICO8 or ARDUBOY : it gets hot rapidly and shuts down abruptly after about 10 minutes when too hot. A 5mn session with PICO8 used 5% battery, [...]

Hey everyone. Sorry for the delay. I can also confirm that, after a lot of playtesting, it behaves much better thermally after replacing the.dtb file. Thanks!

closed 2025-03-04 by christianhaitian