christianhaitian/arkos #1297: Unable to get ArkOS running

gbuffry85, 2025-03-02:
Apologies if this isn't the place to put this Christian but I am having no luck with getting ArkOS to run on my R36s, which was received a few days ago and I am pretty desperate. Really at my wits end. I am not a complete novice - I have customised the SNES mini with ROMS and emulators so I have a basic understanding, although it is very much basic.

I have detailed a bit more below but the summary is that when trying to boot the device I am generally just getting a black screen with faint white vertical lines (almost barcode like) that change, as if the screen is not displaying what it should be properly. On some occasions I have got an orange coloured screen with vertical lines too. I have waited to see if the device will reboot and the software will load properly but it doesn't. Picture links to imgbb are provided below if they help.

DETAILS

I am on a Mac and my R36s was purchased from here...

https://nl.aliexpress.com/item/1005007226765269.html?spm=a2g0o.productlist.main.7.7f2e6cb35uj5rZ&aem_p4p_detail=20250302005958153257802508560016342699&algo_pvid=ee697c55-12a4-442e-95e8-2b5b1e188ec0&algo_exp_id=ee697c55-12a4-442e-95e8-2b5b1e188ec0-3&pdp_ext_f=%7B%22order%22%3A%2224731%22%2C%22eval%22%3A%221%22%7D&pdp_npi=4%40dis%21EUR%21151.73%2140.59%21%21%211122.26%21300.22%21%40%2112000040440720430%21sea%21NL%210%21ABX&curPageLogUid=bGttDCJNnZ4u&utparam-url=scene%3Asearch%7Cquery_from%3A&search_p4p_id=20250302005958153257802508560016342699_3

I purchased a brand new SanDISK Ultra 32GB microSD card (Amazon) and this did not work, so I purchased a Samsung Pro Endurance 32GB (Amazon). Neither work.

I have tried using ApplePiBaker as instructed to image one of the builds onto the device. The first time I got the BOOT and EASYROMS disc but nothing happened on startup. Ever since then I am only getting the BOOT disc with ApplePiBaker. When I go to DiskUtility the second disk is just listed as a generic named disc and I cannot mount it. ApplePiBaker expands the initial IMG to fill the 32GB microSD card as it's flashing.

I know that it says there are issues with BalenaEtcher but I thought I would try this to see if there was any change. This did not automatically expand the IMG file but when it mounts it shows the BOOT and EASYROMS each time. When I try to start the device it is still the same.

I have used a number of builds from yourself as well as the AeolusUX community build but so far nothing has worked. Similarly I have tried replaced the DTB files with those from AelousUX and tech4bot but I can't find a working solution. 

The microSD card that is in the device is working fine and there is no issues with the screen itself as (most of) the games play perfectly. There seems to be no issues with the device itself that I can see as it's all working fine BUT I am obviously concerned about the longevity of the stock microSD card and also want to modify it to add further games.

I had one build where an intro image appeared but then the screen went to a flashing dash in the top left corner but did not do anything else. Unfortunately I did not record which combination this way.

I am a little confused as to which build I should be using for the device but as I say I have tried a number of different combinations. I suppose my main questions are 1) HELP - how do I get this working, 2) is there a way of knowing which combination of IMG / DTB etc I need for this specific device, 3) is this a device issue or ArkOS not populating properly through ApplePi / balena.

Lastly - in a worst case scenario, can I use ApplePi / balena to clone the stock microSD card to a legitimate microSD card and use it that way? And if I cloned it to (for instance) a 128GB can I then modify the ROMS on it - add new ones etc, even if I can't add new systems?

If there is any further information I can add then please let me know - particularly if there is a way of showing what build the device itself is. I am at my wits end trying to work it out so any help would be GREATLY appreciated and I would be eternally grateful. Very happy to donate for any advice to get it working!

IMAGES

https://ibb.co/Kxwn7TJr
https://ibb.co/prkYNDZh
https://ibb.co/JFMv0q9d
https://ibb.co/N2JNHv1w

With stock SD card

https://ibb.co/yBsMM5c5
https://ibb.co/zTQNTcbd
https://ibb.co/7dnQnRxs
https://ibb.co/fVcnVKpZ

christianhaitian, 2025-03-02:
I can not help you with those units as i do not support them. Try the AeolusUX community image  and if that doesn’t work, issues should be entered at https://github.com/AeolusUX/ArkOS-R3XS

closed 2025-03-02 by christianhaitian

gbuffry85, 2025-03-02:
Apologies - when you say units, does this mean the purchased R36 is the wrong one for ArKOS to work on? Thanks!

christianhaitian, 2025-03-02:
I don’t support r36 units