christianhaitian/arkos #1300: RG353M - Shutdown System Revice Instead of Shutting Down; Thememaster Hangs When Attempting Update

Garisaan, 2025-03-04:
### Before submitting your issue or feature request.  DO NOT REQUEST ASSISTANCE FOR PRELOADED IMAGES.  I do not support or condone the use of preloaded ArkOS images.  Preloaded meaning images that come with copyrighted roms preincluded.  Look for support from the creator or supplier of those preloaded images, not from here!  If a ticket is submitted for assistance for such images or if there's an indiciation that a preloaded image is what the support request is for, the issue ticket will be closed with no further response.

1. ***This is not the Stock OS that comes with your device, AmberElec, JelOS, ROCKNIX or UnOfficialOS.***  ArkOS is a different distribution from those systems and will have differing features.  The only somewhat similarity is that they both use EmulationStation as a frontend.  Even with that, the versions of EmulationStation that's used in between ArkOS and these other distributions is different with different capabilities.  If there are features you prefer with the other available distributions, just use those distributions instead.  The choice is yours.

2. Have you reviewed the Frequently Asked Questions section available for your supported device in the wiki?

3. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

4. Have you reviewed the ***known issues*** section available at https://github.com/christianhaitian/arkos/wiki/Known-issues ?

### Describe the Issue (If applicable)

Two issues I've discovered using the latest (as of writing this post) firmware for the Anbernic RG353M device., which is currently using a two SD card setup.

1. Shutting down the device via the Main Menu (Quit -> Shutdown System) does not function as it should, which it instead restarts the device.  The only way I can shut down the device it either plugging in a USB connection to the DC port then shutting down or by performing a hard shutdown by holding down the Power button until it turns off.

2. Thememaster v5.2.0 upon accessing it via Options -> Tools -> Thememaster does not respond to any controls upon receive an update notification ("A new release is available!! Launch update? <Yes, now!> <No, later...>").  Manually downloading the latest update via the Thememaster GitHub or website (v.5.3.0) resolves this hangup issues and functions accordingly.

### How can the issue be reproduced? (If applicable)

See the above section for details.

### What device are you using? (Only the Gameforce Chi, RK2020, OGA 1.0, OGA 1.1, RG351MP, RG351V, RGB10, R35s, RG353V/VS, RG353M, RG503, RGB20 Pro, RGB30, and RK2023 units are officially supported.  Any other device mentioned will most likely result in the issue being closed.)  For assistance with R36s units, please refer to the community image maintained by AeolusUX available [here](https://github.com/AeolusUX/ArkOS-R3XS)

Anbernic RG353M

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen

ArkOS 2.0 (02092025)

### Anything else you'd like to include that may help to review this issue or feature request?

Nothing else to add.

christianhaitian, 2025-03-04:
If you were running ROCKNIX or any other OS that uses the mainline kernel, you need to disconnect the battery on the unit for a few seconds to resolve this issue.  The mainline kernel used in ROCKNIX changes how the power management IC functions that's not compatible with the BSP kernel used in ArkOS.

Garisaan, 2025-03-04:
Thank you!  That resolved the first issue.  As for the Thememaster issue, updating it manually as previously mentioned resolves the hangup from the previous version.

closed 2025-03-04 by Garisaan