christianhaitian/arkos #1301: What is this?

thedicarlos, 2025-03-09:
[image]
When I upgraded to the latest OS, I have this RetroArch overlay that I can't get rid of.  It is on all the emulators.  Please help!

christianhaitian, 2025-03-09:
I can't replicate this issue on my end.  Are you using the image I provided here?

christianhaitian, 2025-03-09:
Oh this is the powkiddy r20s.  Sorry, I don't support this device.  Perhaps try a community image or UnofficialOS.

closed 2025-03-09 by christianhaitian

thedicarlos, 2025-03-09:
I am using the image you provided!

RetroGFX, 2025-03-09:
Christain doesn't officially support this unit. Even tho it may work, the image isn't designed for this device.  Either turn off overlays in the retroarch settings, or you could try a different cfw. There's a few that work

lauren7ino, 2025-03-10:
This is an overlay for playing with touchscreen. RetroArch disable the touchscreen overlay.