christianhaitian/arkos #1303: Zip Support

supermodio64, 2025-03-12:
### Have you reviewed the Frequently Asked Questions section available for your supported device in the wiki?

- [x] I have reviewed the FAQ for my supported device
- [x] I couldn't find the answer for this in the Frequently Asked Quetions section for my supported device

### If your issue is related to being able to run certain emulators, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

- [x] I have reviewed the Emulators and Ports section.
- [x] I couldn't find the answer for this in the Emulators and Ports section.

### Which device are you having this issue with?

RG353V/VS

### Description

I had to edit the es_system.cfg to recognize .zip files for the following:

- wasm4
- arduboy
- uzebox
- n64
- odyssey2

There might be other systems as well. Seems like a quick fix for your next update. Appreciate the continued support. 

### Steps to reproduce the bug

Loaded .zip files that weren’t recognized 

### What version of ArkOS are you on?

Latest

christianhaitian, 2025-03-12:
- [wasm4 core info file](https://github.com/libretro/libretro-super/blob/master/dist/info/wasm4_libretro.info) documentation doesn't state .zip is supported.  

- ardens emulator seems to be the only one that might support .zip but the [documentation (core info file)](https://github.com/libretro/libretro-super/blob/master/dist/info/ardens_libretro.info) doesn't state that's an officially supported extension, [arduous](https://github.com/libretro/libretro-super/blob/master/dist/info/arduous_libretro.info), which is another selectable emulator for arduboy, doesn't support .zip.

- [uzebox core info file](https://github.com/libretro/libretro-super/blob/master/dist/info/uzem_libretro.info) doesn't state .zip is supported 

- n64 already had .zip support already included.  This was added since the 06/24/2024 update which also added .7z support.  I don't know why it was missing from your installation unless you edited the supported extensions entry in your es_systems.cfg file before then in which case the update would not have been able to alter your change.

- [02em core info file](https://github.com/libretro/libretro-super/blob/master/dist/info/o2em_libretro.info) doesn't state .zip is supported

I'm not going to include additional supported extensions unless specifically stated by the emulator or core creators themselves.  If the cores happen to support zip, could be accidental and not purposely designed that could be taken away or broken later unless the developer specifically states the feature is officially supported.

closed 2025-03-12 by christianhaitian

supermodio64, 2025-03-12:
Happy to hear n64 zips are supported. The others seem to work as well so it might confuse users like myself who’ve been playing them compressed but I understand your reservation. Thanks for the consideration and all the work you do.