christianhaitian/arkos #1304: Perfect Overlays | It would be possible to add these overlays to Arkos

furbi963, 2025-03-13:
### Have you reviewed the Frequently Asked Questions section available for your supported device in the wiki?

- [x] I have reviewed the FAQ for my supported device
- [x] I couldn't find the answer for this in the Frequently Asked Quetions section for my supported device

### If your issue is related to being able to run certain emulators, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

- [x] I have reviewed the Emulators and Ports section.
- [x] I couldn't find the answer for this in the Emulators and Ports section.

### Which device are you having this issue with?

Not Listed

### Description

Basically that, add overlays.
For GB, GBA, NGP portable consoles, etc.
Would give a beautiful view to those consoles

Se vería bonito ෆ╹ .̮ ╹ෆ🔪
Very bonito, ése 🤠
https://github.com/mugwomp93/muOS_Customization?tab=readme-ov-file

### Steps to reproduce the bug

Ninguno, solo lo pongo para crear el hilo 🥺✌️

### What version of ArkOS are you on?

la última

christianhaitian, 2025-03-13:
You're welcome to add them yourself but I will not include them by default.

closed 2025-03-13 by christianhaitian

furbi963, 2025-03-13:
Ya los agregue, yo decía para que todo el mundo lo tuviera (っ˘̩╭╮˘̩)っ🔪

I already add them, I said so that everyone had it. (╯︵╰,)✌️

christianhaitian, 2025-03-13:
Anyone else who wants them, can add them themselves as well.  Not everyone cares for overlays like myself.  I like the emulation pure and unfiltered and uncluttered with unneeded extra items.