christianhaitian/arkos #1308: volume control on rg351v

broly9k, 2025-03-31:
### Have you reviewed the Frequently Asked Questions section available for your supported device in the wiki?

- [x] I have reviewed the FAQ for my supported device
- [x] I couldn't find the answer for this in the Frequently Asked Quetions section for my supported device

### If your issue is related to being able to run certain emulators, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

- [x] I have reviewed the Emulators and Ports section.
- [x] I couldn't find the answer for this in the Emulators and Ports section.

### Which device are you having this issue with?

Not Listed

### Description

I'm not sure if its a hardware issue but it was working fine on the stock os. I love the arkos its amazing and runs great, however upon flashing and installation it does not seem to let me control the device volume from the side buttons on the device (nothing happens at all with these buttons) . I'm constantly having to go into settings and turn down the "gain" with every game startup. I'm not sure if this could be a simple fix on my part or simply a bug? thank you! you do an awesome job!!!

### Expected behavior

the overall volume to turn down and remain that way when using the side buttons on device for volume control.

### Steps to reproduce the bug

power device on
try using side buttons to change device volume

### What version of ArkOS are you on?

03302025

christianhaitian, 2025-03-31:
Volume control works fine on my 351v install.   Have you tried using the fix global hotkeys tool in the options advanced section?

broly9k, 2025-03-31:
It says are you sure you want to restart global hot keys, is that okay?

broly9k, 2025-03-31:
Device is freezing on that screen every time I load it

christianhaitian, 2025-03-31:
You should be able to press a to continue

broly9k, 2025-03-31:
When it pops up on that screen it freezes when I click a

broly9k, 2025-03-31:
> I can't replicate this issue at all on my end. Did you have to do anything
> special to load ArkOS on your unit? Make any dtb changes?

I did have to replace a file due to it booting up weirdly with pixelated
screen and a black bar on the side

broly9k, 2025-03-31:
Nevermind that was when I tried running amberelec, but no I didn’t replace
any files with arkos

christianhaitian, 2025-04-01:
What brand sd card are you using?  What software did you use to apply the image to the sd card?

christianhaitian, 2025-04-01:
What is the brand of the sd card?

broly9k, 2025-04-01:
onn

christianhaitian, 2025-04-01:
Try a Sandisk or Samsung sd card.

christianhaitian, 2025-04-03:
Any update on this?  Have you tried a Sandisk or Samsung sd card to see if the issue happens on one of those cards as well?

broly9k, 2025-04-03:
I have not had the chance to buy one but I will today, but I can not flash
it until I get a new laptop my windows laptop broke, I have a Chromebook
but I don’t believe you can flash with those. But the system runs great,
just used myrient to load over 1500 roms using packs, everything runs fine.
I just go to system setting and adjust volume if needed. But the updates
install smooth and eve the thing else, just that hot key isssue, all
buttons are mapped correctly to each rom, the volume button isn’t a major
issue, but I will be installing arkos on my new device (if it runs arkos)
it’s a cubeXX

AdminVin, 2025-04-15:
Hey there!

Issue present on my RG351V, that the volume up and volume down do not work. I confirmed they weren't being detected in the button test. Only two buttons that do not present output when pressed.

I attempted to run "Fix Global HotKeys", but pressing A freezes the device.

SD Card: https://www.amazon.com/dp/B07YQ5M4V4

I did use the official RG351V DTB file to fix the screen issue.

- Downloaded latest version from Anbernic 
- Extracted rk3326-anbernic-st7703.dtv
- Renamed rk3326-anbernic-st7703.dtv renamed to rk3326-rg351v-linux.dtb
- Wrote the latest version of ArkOS to the SD card with Win32 Disk Imager
- Used DiskGenius to access boot partition and replace the file
- Device booted with the screen working

broly9k, 2025-04-15:
Can I simply update arkOS on the device and that fixes it or do I need to
reflash?

AdminVin, 2025-04-16:
Update: 

**Global Hotfix - Not working**
- I used ChatGPT to run it it through SSH
- sudo bash "/opt/system/Advanced/Fix Global Hotkeys.sh"
- Output:
open: No such file or directory
open: No such file or directory
open: No such file or directory
open: No such file or directory
open: No such file or directory
^C
ark@rg351v:/$ ^C

- cat "/opt/system/Advanced/Fix Global Hotkeys.sh"
- Output:
sudo systemctl stop oga_events
sudo systemctl disable oga_events
sudo cp -f -v /etc/systemd/system/oga_events.service.bak /etc/systemd/system/oga_events.service
sudo systemctl daemon-reload
sudo systemctl start oga_events
sudo systemctl enable oga_events
- After running that, volume buttons did not work.

**ChatGPT thinks an issue with the boot not initializing the buttons properly**
- strings /proc/device-tree/compatible
- Output:
ark@rg351v:~$ strings /proc/device-tree/compatible
rockchip,rk3326-odroidgo2-linux
rockchip,rk3326
ark@rg351v:~$

- ChatGPT thinks the issue is
"You’re booting with the wrong DTB: rockchip,rk3326-odroidgo2-linux
This means ArkOS is still loading: rk3326-odroidgo2-linux.dtb"

I don't know Linux very well and work in Windows primarily. Hope this helps!

closed 2025-06-18 by christianhaitian