christianhaitian/arkos #1309: Contribution to Emulation Station (release date + manufacturer + hardware type)

lauren7ino, 2025-04-01:
### Have you reviewed the Frequently Asked Questions section available for your supported device in the wiki?

- [x] I have reviewed the FAQ for my supported device
- [x] I couldn't find the answer for this in the Frequently Asked Quetions section for my supported device

### If your issue is related to being able to run certain emulators, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

- [x] I have reviewed the Emulators and Ports section.
- [x] I couldn't find the answer for this in the Emulators and Ports section.

### Which device are you having this issue with?

Not Listed

### Description

Hi, greetings from Portugal

I've added metadata to the current (yesterday) ArkOS es_systems.cfg, so that Emulation Station filters can run on release date, manufacturer and hardware type.

I hope you find this information useful. I made mostly for fun and self learning.

PS: One of the systems you may want to review is RetroArch. I input 1971 since it runs hardware from 1971 (first Arcades), but maybe we should remove the date. I don't know. You decide. Thanks

I'm not of a GitHub expert, so I leave it here for you to incorporate if you wish.

[image]

[es_systems.cfg.zip](https://github.com/user-attachments/files/19550609/es_systems.cfg.zip)

### Expected behavior

.

### Steps to reproduce the bug

.

### What version of ArkOS are you on?

03292025

christianhaitian, 2025-04-01:
Thank you.  I've thought about adding this feature but did not because it would require wiping any customizations that some do to their es_systems.cfg.  Since I back that information up before applying such updates, I think I will include this.  I will review and make any modifications I feel is necessary and include it with the next update.  Thanks again for this contribution.

lauren7ino, 2025-04-01:
You're welcome! Looking forward to the update!

lauren7ino, 2025-04-01:
I created systems.json from a CSV sheet I have.

I'm giving you this Python script I used once to inject the systems.json into the existing (ArkOS) es_system.cfg

[python_script.zip](https://github.com/user-attachments/files/19555161/python_script.zip)

Dumps a missing_systems.json for those not found.
The end xml a bit unformatted but still syntax correct.

It's like a es_systems patcher.

closed 2025-04-30 by christianhaitian