christianhaitian/arkos #1311: Power button works, but doesn't lock the screen

Plaque-fcc, 2025-04-14:
### Have you reviewed the Frequently Asked Questions section available for your supported device in the wiki?

- [x] I have reviewed the FAQ for my supported device
- [x] I couldn't find the answer for this in the Frequently Asked Quetions section for my supported device

### If your issue is related to being able to run certain emulators, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

- [x] I have reviewed the Emulators and Ports section.
- [x] I couldn't find the answer for this in the Emulators and Ports section.

### Which device are you having this issue with?

Not Listed

### Description

I am using ArkOS 2.0 (03302025)(AeUX), latest release on my R36S.
The **power** button is supposed to power on/off the device, and tapping it should **lock the screen**, when on.

√ Fresh install.
√ Three different microSD cards (64GB and 128GB) tested.
√ Didn't interrupt the first boot process, merely let it do its thing and reboot.
√ On release ArkOS 2.0 (02092025)(AeUX) the power button works as expected: lets me power on (or force power off, if held for long) the device, or lock the screen.
√ When I install 02092025 and _then_ upgrade it to the latest release, the button works as intended.

So that's what I did, in the end. Hope this release will get pulled out and reissued with fixes.

And thank you for your amazing work.

### Expected behavior

The power button used to lock the screen when tapped.

### Steps to reproduce the bug

1. Write 03302025 release image onto a microSD card.
2. Insert the card into your R3xS handheld and let it set it up and reboot.
3. Try locking the screen by tapping *power* button — it will likely not lock the screen.

### What version of ArkOS are you on?

03302025

christianhaitian, 2025-04-14:
You reported this in the wrong github.  Please report the the community distro github.

closed 2025-04-14 by christianhaitian

Plaque-fcc, 2025-04-14:
> You reported this in the wrong github. Please report the the community distro github.

You're absolutely correct, and if I took my time to find the correct repo, I'd even save me some time.

And — thank you again.