christianhaitian/arkos #1313: DOSBox Pure crashing on custom keyboard bindingon RGB30

mikhailter, 2025-04-14:
### Have you reviewed the Frequently Asked Questions section available for your supported device in the wiki?

- [x] I have reviewed the FAQ for my supported device
- [x] I couldn't find the answer for this in the Frequently Asked Quetions section for my supported device

### If your issue is related to being able to run certain emulators, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

- [x] I have reviewed the Emulators and Ports section.
- [x] I couldn't find the answer for this in the Emulators and Ports section.

### Which device are you having this issue with?

Not Listed

### Description

Can't choose Custom Keyboard Bindiing becase Retroarch is crashing with this option

### Steps to reproduce the bug

1. Start DOS game with DOSBox Pure core
2. Open Retroarch menu with Select + X
3. Goto: Controls > Port 1 Controls  > Device Type, and then select "Custom Keyboard Bindiings" and the core is crashing

### What version of ArkOS are you on?

03292025

christianhaitian, 2025-04-16:
I recently updated this core, try updating it from the retroarch repo and see if that resolves the issue.

mikhailter, 2025-04-17:
This core is working fine without crash:

[image]

btw, I updated accidentally all the cores, is there a simpe way to revert all cores to latest arks verions's ?

closed 2025-04-17 by christianhaitian