christianhaitian/arkos #1314: RG351V - Volume Controls

AdminVin, 2025-04-16:
### Have you reviewed the Frequently Asked Questions section available for your supported device in the wiki?

- [x] I have reviewed the FAQ for my supported device
- [x] I couldn't find the answer for this in the Frequently Asked Quetions section for my supported device

### If your issue is related to being able to run certain emulators, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

- [x] I have reviewed the Emulators and Ports section.
- [x] I couldn't find the answer for this in the Emulators and Ports section.

### Which device are you having this issue with?

RG351V

### Description

Hey there!

Issue present on my RG351V, that the volume up and volume down do not work. I confirmed they weren't being detected in the button test. Only two buttons that do not present output when pressed.

SD Card: https://www.amazon.com/dp/B07YQ5M4V4

**Only Modification - Screen issue**

I did use the official RG351V DTB file to fix the screen issue. When I flashed it initially it was a white screen.

- Downloaded latest version from Anbernic 
- Extracted rk3326-anbernic-st7703.dtb
- Renamed rk3326-anbernic-st7703.dtb renamed to rk3326-rg351v-linux.dtb
- Wrote the latest version of ArkOS to the SD card with Win32 Disk Imager
- Used DiskGenius to access boot partition on the SD card and replace the file.
- Device booted with the screen working.

**Global Hotfix - Not working**
- I used ChatGPT to run it it through SSH
- sudo bash "/opt/system/Advanced/Fix Global Hotkeys.sh"
- Output:
open: No such file or directory
open: No such file or directory
open: No such file or directory
open: No such file or directory
open: No such file or directory
^C
ark@rg351v:/$ ^C

- cat "/opt/system/Advanced/Fix Global Hotkeys.sh"
- Output:
sudo systemctl stop oga_events
sudo systemctl disable oga_events
sudo cp -f -v /etc/systemd/system/oga_events.service.bak /etc/systemd/system/oga_events.service
sudo systemctl daemon-reload
sudo systemctl start oga_events
sudo systemctl enable oga_events
- After running that, volume buttons did not work.

**ChatGPT thinks an issue with the boot not initializing the buttons properly**
- strings /proc/device-tree/compatible
- Output:
ark@rg351v:~$ strings /proc/device-tree/compatible
rockchip,rk3326-odroidgo2-linux
rockchip,rk3326
ark@rg351v:~$

- ChatGPT thinks the issue is
"You’re booting with the wrong DTB: rockchip,rk3326-odroidgo2-linux
This means ArkOS is still loading: rk3326-odroidgo2-linux.dtb"

I work in Windows primarily, so I did some troubleshooting to see if anything sticks out for you. When you get a chance any advise would be appreciated. Thank you!

### Expected behavior

Volume should increase or decrease when the buttons are pushed

### Steps to reproduce the bug

Boot device and press volume buttons

### What version of ArkOS are you on?

03302025

christianhaitian, 2025-04-16:
Unfortunately, this is not an issue I'm experiencing on my end on my unit so I don't know how to address this.  I may close this issue in the near future as I simply don't support the new RG351V units as I don't have one I test against.

AdminVin, 2025-04-16:
Understood. Does this make sense though? 

"ChatGPT thinks an issue with the boot not initializing the buttons properly

strings /proc/device-tree/compatible

Output:
ark@rg351v:$ strings /proc/device-tree/compatible
rockchip,rk3326-odroidgo2-linux
rockchip,rk3326
ark@rg351v:$

ChatGPT thinks the issue is
"You’re booting with the wrong DTB: rockchip,rk3326-odroidgo2-linux
This means ArkOS is still loading: rk3326-odroidgo2-linux.dtb"

Any idea the easiest way to accomplish this, to test?

christianhaitian, 2025-04-16:
chatgpt is wrong as if that was the dtb you were loading with, the unit wouldn't boot at all.

AdminVin, 2025-04-16:
Sounds good, appreciate the insight! I'll keep messing with it and advise if I can get a fix going for the newer style units.

closed 2025-04-22 by christianhaitian