christianhaitian/arkos #1317: Screen Saver

lauren7ino, 2025-04-18:
### Have you reviewed the Frequently Asked Questions section available for your supported device in the wiki?

- [x] I have reviewed the FAQ for my supported device
- [x] I couldn't find the answer for this in the Frequently Asked Quetions section for my supported device

### If your issue is related to being able to run certain emulators, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

- [x] I have reviewed the Emulators and Ports section.
- [x] I couldn't find the answer for this in the Emulators and Ports section.

### Which device are you having this issue with?

OGA 1.1/RGB10/V10

### Description

Hi, this is not a bug, just a suggestion:
In DIM screen saver mode, the RGB30 screen dims too little, almost not saving any battery.
It seems to reduce the brightness by about half, but I expected it to reduce to 5% or less.
Thanks.

### Expected behavior

.

### Steps to reproduce the bug

.

### What version of ArkOS are you on?

20250330

christianhaitian, 2025-04-18:
Of you can figure out how to achieve that in the ES code used [here](https://github.com/christianhaitian/EmulationStation-fcamod/tree/503noTTS) and submit the change as a PR, I'd be happy to review it as a possible addition.

closed 2025-04-22 by christianhaitian

lauren7ino, 2025-04-29:
Sorry, I don’t have the time to help right now. Thank you for understanding.