christianhaitian/arkos #1320: Retrorun32 flycast_rumble and windows CE games

GazousGit, 2025-04-21:
### Have you reviewed the Frequently Asked Questions section available for your supported device in the wiki?

- [x] I have reviewed the FAQ for my supported device
- [x] I couldn't find the answer for this in the Frequently Asked Quetions section for my supported device

### If your issue is related to being able to run certain emulators, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

- [x] I have reviewed the Emulators and Ports section.
- [x] I couldn't find the answer for this in the Emulators and Ports section.

### Which device are you having this issue with?

R35S

### Description

Hello Christian,

I'm sorry if posting a new issue is not the best way to get around this but I have some questions.

I'm using a R36S device (but using your fork of ArkOS (latest is end of march fork I think)

My question revolves a bit more about Retrorun32.

I'm using the combination of retrorun32 and flycast_rumble which does wonders to be fair.
However when I try to get Resident Evil 2 Dreamcast, it always hang on the Windows CE screen (right after the boot logo of the dreamcast)

Every other emulator get past it but they don't offer great performance of course...

I see that flycast has an option that might resolve the issue:
Force Windows CE Mode [flycast_force_windows_ce_modee] (disabled|enabled)
Enable full MMU (Memory Management Unit) emulation and other settings for Windows CE games.

The issue is, I have no idea how I can get it using retrorun32 and flycast_rumble (I can access the R36S with SCP/putty but I don't know where the conf file is located and (at best) if I could create files per game if necesary.

Thank you for your time.

### Steps to reproduce the bug

1.[Set retrorun32 with flycast_rumble for dreamcast]
2.[Try Resident evil 2 game (which has Windows CE validation step]
3.[Watch endlesssly the windows CE screen, cause it kinda freeze there, accessing flycast_menu is possible but if you "resume" it will be frozen]

### What version of ArkOS are you on?

03302025

GazousGit, 2025-04-21:
Additional Info:
I found a retrorun config file in /home/ark/.config

I tried the settings:
reicast_force_wince
flycast_force_wince

But nothing change

Also Resident Evil 3 (which also require WinCE seems to work)

christianhaitian, 2025-04-22:
I have no idea for this game.  If it doesn’t work for the core within retroarch, it most like won’t work in retrorun either.  If it does work in retroarch, it’s probably an incompatibility with retrorun.

closed 2025-04-22 by christianhaitian

GazousGit, 2025-04-22:
it actually does work on retroarch.
I just can't find why the resident evil 2 gets a sort of "disabled" WinCE when it does not for resident evil 3.
Can you still help me to understand why the flycast32_rumble core would ouput that specific line

Flycast < [INFO] nullDC.cpp:196 N[BOOT]: [Hack]: Disabling WinCE virtual memory.