christianhaitian/arkos #1325: Crashing due to python event at random randomly

manzaloros, 2025-04-24:
### Have you reviewed the Frequently Asked Questions section available for your supported device in the wiki?

- [x] I have reviewed the FAQ for my supported device
- [x] I couldn't find the answer for this in the Frequently Asked Quetions section for my supported device

### If your issue is related to being able to run certain emulators, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

- [x] I have reviewed the Emulators and Ports section.
- [x] I couldn't find the answer for this in the Emulators and Ports section.

### Which device are you having this issue with?

RGB30

### Description

I haven’t tested with all emulators, but with game boy color on retro arch, the screen will, at random, go dark, and a small pop-up in the bottom left corner will briefly mention some python error.

The error flashes so quickly that I can’t see the full text, but it might say something like event overloaded.

I can’t be sure, but the error looks like it might be coming from retro arch?

### Expected behavior

I expect to play emulators with no random crash crashes.

### Steps to reproduce the bug

1. Play a game, possibly game boy, color, on retro arch for around 10 minutes.
2. Observed the screen go black with a python error.
3. Observed that the only way to recover from the state is to hit the hardware reset button on the device.

### What version of ArkOS are you on?

Latest

christianhaitian, 2025-04-24:
Do you happen to have Auto Suspend Timeout enabled in the emulationstation>Advanced Settings menu?

manzaloros, 2025-04-25:
Hmm, I do have AutoSuspendTimeout set to 15 minutes in the settings.

I don't remember setting this value, though.

It seems like that setting should only auto suspend if there's no activity for 15minutes, is that not correct?

christianhaitian, 2025-04-25:
That is correct however it's been misbehaving so I need to find out why. Try disabling it for now.

closed 2025-04-26 by manzaloros