christianhaitian/arkos #1326: Retrorun quit button

GazousGit, 2025-04-27:
### Have you reviewed the Frequently Asked Questions section available for your supported device in the wiki?

- [x] I have reviewed the FAQ for my supported device
- [x] I couldn't find the answer for this in the Frequently Asked Quetions section for my supported device

### If your issue is related to being able to run certain emulators, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

- [x] I have reviewed the Emulators and Ports section.
- [x] I couldn't find the answer for this in the Emulators and Ports section.

### Which device are you having this issue with?

R35S

### Description

Hello,

I wondered if it was possible to change the mapping on retrorun for the exit state ?

I'm bothered a lot by the fact it's on the start button on my device, many games will require you to press start several times on the menus or even in game to do stuff and because it's the exit command for retrorun you always have to wait for the "Press again to quit" pop-up to go away before you can proceed and press start again.

Best Regards.

### Steps to reproduce the bug

It's not a "bug" of course but a request (or any info that would help me change the input (even if I have to recompile retrorun, I'd gladly do so)

### What version of ArkOS are you on?

03302025

christianhaitian, 2025-04-27:
This issue should be fixed with the latest version of retrorun.  Select+Start should be what's required to exit exit the emulator via hotkey.  Since you're using a R35S, it could be an issue related to that unit not being recognized as a Rg351MP clone.  You can check the dreamcast.sh script in /usr/loca/bin to see how the detection is done.

GazousGit, 2025-04-29:
Alright thanks for the infos Christian, as I mentionned in an older issue about flycast, I'm actually on R36S.
I know you don't support them and I'm actually on the community version of ArkOS for R36S so I guess I'll have to wait for this one to catch up with your latest update.

However, if this is actually indeed linked to the device itself (I kind of doubt it because previous version of ArkOS did not had this behavior to quit retrorun) I may dig a bit into building retrorun to temper with buttons and make a fork dedicated for R36S :)