christianhaitian/arkos #1330: RG351V - Brightness hotkey minimum 15%

05warmupsdelayer, 2025-05-03:
### Have you reviewed the Frequently Asked Questions section available for your supported device in the wiki?

- [x] I have reviewed the FAQ for my supported device
- [x] I couldn't find the answer for this in the Frequently Asked Quetions section for my supported device

### If your issue is related to being able to run certain emulators, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

- [x] I have reviewed the Emulators and Ports section.
- [x] I couldn't find the answer for this in the Emulators and Ports section.

### Which device are you having this issue with?

RG351V

### Description

Brightness hotkeys (both dpad / volume) minimum value is 15% - emulation station’s brightness control lets me go to 1%. 

This is really annoying at night when playing a game and screen is too bright. 

### Expected behavior

I expect full range of brightness control using hot keys, including the lower 15%

### Steps to reproduce the bug

Try and lower brightness using hot keys

### What version of ArkOS are you on?

03292025

christianhaitian, 2025-05-03:
It goes as low as 5% via the global hotkeys.  Not sure why it's being limited to 5 though.  Not something I will spend time figuring out at the moment though.  You're welcome to review the source for the global hotkey daemon here: https://github.com/christianhaitian/ogage/blob/rg351v/src/main.rs, change it, and rebuild it yourself.  Brightness controls start at this line: https://github.com/christianhaitian/ogage/blob/592cdbeffc2bdd4b549de96bb55d93df01241dee/src/main.rs#L59

05warmupsdelayer, 2025-06-17:
Line 63:  Command::new("brightnessctl").args(&["**-n15**","s","2%-"]).output().expect("Failed to execute brightnessctl");

This part, the -n15 is that the min %?

05warmupsdelayer, 2025-06-18:
Thanks, I was able to set to 1% and build on RG351V

https://github.com/05warmupsdelayer/ogage/releases/tag/ogage-rg351