christianhaitian/arkos #1331: After 4/30/25 update PortMaster and ThemeMaster stopped working

DingirAn, 2025-05-08:
### Have you reviewed the Frequently Asked Questions section available for your supported device in the wiki?

- [x] I have reviewed the FAQ for my supported device
- [x] I couldn't find the answer for this in the Frequently Asked Quetions section for my supported device

### If your issue is related to being able to run certain emulators, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

- [x] I have reviewed the Emulators and Ports section.
- [x] I couldn't find the answer for this in the Emulators and Ports section.

### Which device are you having this issue with?

RG351MP

### Description

After the update, aside from the issue with the Roms not showing (I tried the SD2 to SD1 to SD2 switching which solved the ROMS issue), PortMaster and ThemeMaster disappeared from Tools and no port start up anymore. 

### Steps to reproduce the bug

1. Update to 4/30/25 
2. That’s it. 

### What version of ArkOS are you on?

03302025, had to reflash and rollback.

christianhaitian, 2025-05-09:
This issue should be resolved now if you update again

DingirAn, 2025-05-10:
It seems it has been solved, thanks a lot!

closed 2025-05-10 by DingirAn