christianhaitian/arkos #1332: Max Payne GBA doesn't save progress

Ulyapka, 2025-05-09:
### Have you reviewed the Frequently Asked Questions section available for your supported device in the wiki?

- [x] I have reviewed the FAQ for my supported device
- [x] I couldn't find the answer for this in the Frequently Asked Quetions section for my supported device

### If your issue is related to being able to run certain emulators, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

- [x] I have reviewed the Emulators and Ports section.
- [x] I couldn't find the answer for this in the Emulators and Ports section.

### Which device are you having this issue with?

OGA 1.1/RGB10/V10

### Description

When the game prompts me to save, the save happens, but before the game restarts. When the game is turned off and restarted, everything disappears. 

### Steps to reproduce the bug

1. Run Max Payne GBA
2. Complete first chapter
3. Save game
4. Restart game

### What version of ArkOS are you on?

03292025

christianhaitian, 2025-05-09:
What process are you using to exit the game?

Ulyapka, 2025-05-09:
Double-click the “start + select” buttons

Ulyapka, 2025-06-18:
I'll try reproducing the problem again on a fresh system update and test on other systems, then come back with a detailed report. 
If this problem is present in one game, it may occur in other games as well. I want to avoid similar problems in the future, that's why I've opened this issue.

Ulyapka, 2025-06-22:
Apparently the problem was with the ROM, as it's the same problem on other systems.

closed 2025-06-22 by Ulyapka