christianhaitian/arkos #1336: Pico-8 mouse mode support

deliveryturtle, 2025-05-23:
### Have you reviewed the Frequently Asked Questions section available for your supported device in the wiki?

- [x] I have reviewed the FAQ for my supported device
- [x] I couldn't find the answer for this in the Frequently Asked Quetions section for my supported device

### If your issue is related to being able to run certain emulators, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

- [x] I have reviewed the Emulators and Ports section.
- [x] I couldn't find the answer for this in the Emulators and Ports section.

### Which device are you having this issue with?

RGB30

### Description

Unable to play certain Pico-8 which require mouse input, on all cores (Paid Pico-8/ Fake-08/ Retroarch).

### Expected behavior

Pressing L2 will switch to mouse mode. Being able to control cursor with joystick and clicking with buttons. (This seems to be the way how other retro handheld system handle it. [Link](https://www.reddit.com/r/MiyooMini/comments/1c4eca4/mouse_mode_on_pico_8_games/) )

### Steps to reproduce the bug

Opening a mouse-only game such as Solitomb using pico-8 core.
Pressing L2.
Not switching to mouse mode.

### What version of ArkOS are you on?

04302025

christianhaitian, 2025-05-27:
Only way I see this being possible with the standalone emulators is by using some kind of emulator like gptokeyb.  This is not something that will be high on my list to tackle but I may look into this at some point.

deliveryturtle, 2025-05-29:
Thanks for replying! No problem. Thanks for all the great work.

bodnalev, 2025-07-12:
Thank you @christianhaitian for making and maintaining arkos!
I was also wondering about this functionality today (for the exact same reason, to play solitomb and slipways lol). I managed to make it work on my rgb30 device using xboxdrv. I hope it helps implement this properly so everyone can use it, but if someone is impatient this fairly simple tutorial should work (I could only test it on rgb30):

0. ssh into the device using the info from [here](https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---CHI#q-how-do-i-ssh-into-ArkOS)
1. Get `xboxdrv` using
```
sudo apt-get update
sudo apt-get install xboxdrv
```
2. Create the file `xboxdrv-mouse.ini` using the command:```
sudo nano xboxdrv-mouse.ini
```
3. Paste the following there. Note, the 6 and -6 are sensitivity, it worked well for me, but you can adjust. Also I mapped the shoulder buttons to the left and right mouse. Not sure how intuitive that is for everyone.
```
[xboxdrv]
evdev=/dev/input/event3
silent=true
device-name="joystick-2-mouse"

[evdev-absmap]
ABS_RX=x2
ABS_RY=y2

[evdev-keymap]
BTN_TL=lt
BTN_TR=rt

[ui-axismap]
x2=REL_X:6
y2=REL_Y:-6

[ui-buttonmap]
lt=BTN_LEFT
rt=BTN_RIGHT
```
4. Then save and exit (Ctrl+O, Enter, Ctrl+C)
5. Create file `/etc/systemd/system/xboxdrv-mouse.service` using the command:
```
sudo nano /etc/systemd/system/xboxdrv-mouse.service
```
6. Paste the following there.
```
[Unit]
Description=xboxdrv mouse service
After=systemd-udev-settle.service

[Service]
Type=simple
User=root
ExecStart=/usr/bin/xboxdrv --silent --evdev-no-grab --config /home/ark/xboxdrv-mouse.ini
Restart=always
RestartSec=5

[Install]
WantedBy=multi-user.target
```
7. Save and exit (Ctrl+O, Enter, Ctrl+C)
8. Start the service using the commands:
```
sudo systemctl daemon-reload
sudo systemctl enable xboxdrv-mouse.service
sudo systemctl start xboxdrv-mouse.service
```
9. Restart and enjoy!

It seems that there is a bug in xboxdrv that it waits 60 second for no reason. I didn't want to go down that rabbit hole, but it seems like there is a fix [here](https://github.com/xboxdrv/xboxdrv/pull/214). The apt-get installed 0.8.8 for me, this could be a bit outdated, not sure.

bodnalev, 2025-07-13:
Quick update, I compiled a version of xboxdrv that has the patch mentioned on the linked [page](https://github.com/xboxdrv/xboxdrv/pull/214) on an ARM64 docker.  It works well on my rgb30, so if anyone else wants to use this file then I have it attached below (idk why .deb attachment is not allowed).

[xboxdrv_0.8.8-1_arm64.zip](https://github.com/user-attachments/files/21205116/xboxdrv_0.8.8-1_arm64.zip)

@christianhaitian if you would like to compile xboxdrv (perhaps for a different architecture) I used the following build recipe:
```
FROM --platform=linux/arm64 arm64v8/ubuntu:19.10
ENV DEBIAN_FRONTEND=noninteractive
RUN \
    echo "deb http://old-releases.ubuntu.com/ubuntu eoan main universe" > /etc/apt/sources.list && \
    echo "deb-src http://old-releases.ubuntu.com/ubuntu eoan main universe" >> /etc/apt/sources.list && \
    echo "deb http://old-releases.ubuntu.com/ubuntu eoan-updates main universe" >> /etc/apt/sources.list && \
    echo "deb-src http://old-releases.ubuntu.com/ubuntu eoan-updates main universe" >> /etc/apt/sources.list && \
    echo "deb http://old-releases.ubuntu.com/ubuntu eoan-security main universe" >> /etc/apt/sources.list && \
    echo "deb-src http://old-releases.ubuntu.com/ubuntu eoan-security main universe" >> /etc/apt/sources.list && \
    apt-get update && \
    apt-get install -y build-essential devscripts && \
    apt-get build-dep -y xboxdrv
WORKDIR /build
RUN apt-get source xboxdrv
WORKDIR /build/xboxdrv-0.8.8
RUN sed -i 's|libusb_handle_events(NULL);|{ struct timeval to; to.tv_sec = 0; to.tv_usec = 0; libusb_handle_events_timeout_completed(NULL, \&to, NULL); }|' src/usb_gsource.cpp
RUN debuild -us -uc -b
```