christianhaitian/arkos #1341: Adding a new script under tools and advanced

cahsobo, 2025-06-02:
### Have you reviewed the Frequently Asked Questions section available for your supported device in the wiki?

- [x] I have reviewed the FAQ for my supported device
- [x] I couldn't find the answer for this in the Frequently Asked Quetions section for my supported device

### If your issue is related to being able to run certain emulators, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

- [x] I have reviewed the Emulators and Ports section.
- [x] I couldn't find the answer for this in the Emulators and Ports section.

### Which device are you having this issue with?

RK2023

### Description

Where can I find a guide to add a new script under OPTIONS > Tools and OPTIONS > Advanced?
I couldn't find it neither in the ArkOS wiki nor EmulationStation github page.

### Steps to reproduce the bug

I need to customize/add some shell script below Options Menu

### What version of ArkOS are you on?

05242024

christianhaitian, 2025-06-02:
There’s no guide for that. For Options>Tools, just copy the script to the Tools folder in the easyroms partition. For Options>Advanced, just copy the script to /opt/system/Advanced via ssh or sftp.

cahsobo, 2025-06-02:
> There’s no guide for that. For Options>Tools, just copy the script to the Tools folder in the easyroms partition. For Options>Advanced, just copy the script to /opt/system/Advanced via ssh or sftp.

OMG. Thank You. I tried to copy a script from /opt/systems to /opt/systems/Tools and it works. I guess I was just overthinking.