christianhaitian/arkos #1343: Rg353m volume adjusting issues

wolff889, 2025-06-09:
### Have you reviewed the Frequently Asked Questions section available for your supported device in the wiki?

- [x] I have reviewed the FAQ for my supported device
- [x] I couldn't find the answer for this in the Frequently Asked Quetions section for my supported device

### If your issue is related to being able to run certain emulators, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

- [x] I have reviewed the Emulators and Ports section.
- [x] I couldn't find the answer for this in the Emulators and Ports section.

### Which device are you having this issue with?

RG353M

### Description

I can seem to adjust volume on the unit. I can on android side. I noticed on main menu (start button press) at the bottom “snd” is “n/a” instead of the usual number percent. I tried going into sound settings and adjusting volume there, when I adjust the volume there the volume slider appears , but when I go and check, for example letting a preview video play the sound level doesn’t actually change. /

[image]

### Expected behavior

Usually the sound level shows up at the bottom of main menu and volume buttons are able to adjust volume. 

### Steps to reproduce the bug

I’ve powered off the unit, powered back on problem persists. 

### What version of ArkOS are you on?

Arkos 2.0 (05312025)

wolff889, 2025-06-09:
I tried choosing “fix audio” from the “scripts/setup” section and that seems to have fixed it the issues I was having.

wolff889, 2025-06-09:
Ok while that fixed the issues, I’ve pinpointed that the culprit is “render96ex” from portmaster it causes the issue so far everytime it’s loaded. Volume works as expected until I start render96ex, the volume seems to shoot to max and I can no longer adjust it, once I exit the game I have to run the “fix audio” option to fix it.

christianhaitian, 2025-06-09:
Report the issue with PortMaster for that game.

closed 2025-06-09 by christianhaitian