christianhaitian/arkos #1344: [RGB20PRO] [EmulationStation] Getting out of sleep while a video preview is playing crashes the device

lebrunregis, 2025-06-09:
### Have you reviewed the Frequently Asked Questions section available for your supported device in the wiki?

- [x] I have reviewed the FAQ for my supported device
- [x] I couldn't find the answer for this in the Frequently Asked Quetions section for my supported device

### If your issue is related to being able to run certain emulators, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

- [x] I have reviewed the Emulators and Ports section.
- [x] I couldn't find the answer for this in the Emulators and Ports section.

### Which device are you having this issue with?

RGB20Pro

### Description

On emulationstation, getting out of sleep mode while a video is playing crashes the device.

### Steps to reproduce the bug

1. Hover a game with a video preview
2. Press the power button to enter sleep mode
3. Press the button again to get out of sleep mode
4. The first frame to be loaded will look corrupted and the device will crash

### What version of ArkOS are you on?

05312025

christianhaitian, 2025-06-09:
This is a a known emulationstation bug.  What should happen is when the device is awaken from sleep if a video was playing, emulationstation should automatically restart.  is this not happening?

lebrunregis, 2025-06-09:
I've seen the reload happening, but this doesn't always happen.