christianhaitian/arkos #1348: Add a Menu About the Owner

cahsobo, 2025-06-16:
### Have you reviewed the Frequently Asked Questions section available for your supported device in the wiki?

- [x] I have reviewed the FAQ for my supported device
- [x] I couldn't find the answer for this in the Frequently Asked Quetions section for my supported device

### If your issue is related to being able to run certain emulators, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

- [x] I have reviewed the Emulators and Ports section.
- [x] I couldn't find the answer for this in the Emulators and Ports section.

### Which device are you having this issue with?

OGA 1.1/RGB10/V10

### Description

I need to add a "About the Owner" menu. Where it displays picture, phone number and address of the owner. I think it will come in handy if the handheld is lost.

### Expected behavior

The purpose is quite simple. If a good samaritan find my lost handheld, they can immediately contact me.

### Steps to reproduce the bug

I have tried editing menu, tried image viewer app to displays personal details such as viu, feh, magick, w3m. none of them worked. 

### What version of ArkOS are you on?

05312025

christianhaitian, 2025-06-16:
I usually use ffplay for displaying images. Why not just place a label with these details on the back of your unit?

cahsobo, 2025-06-16:
I've seen inside the **Controller Tester** script and tried to replicate the hotkey script but failed. Where can I read documentation regarding this?

Or can you give me an example on this?

I want to be able to exit ffplay by pressing any button. Or automatically exit at certain seconds.

christianhaitian, 2025-06-18:
Did you put exit after the sleep command?  /usr/local/bin for buttnmon.sh

christianhaitian, 2025-06-20:
Try the following

```
 ffplay new.jpg &
 PROC=$!
 (sleep 8; kill -9 $PROC)
exit
 ```

You can make use of gptokeyb for mapping face buttons to keyboard keys.  The executable is available in `/opt/inttools`.  
Documentation for it is available here: https://github.com/EmuELEC/gptokeyb
You can look at `/usr/bin/msgbox` for an idea on how to use it.

I still believe the most practical solution is to just use a label on the back of the unit. ¯\_(ツ)_/¯