christianhaitian/arkos #1353: How to compile and package Source code (tar. gz) into img

johy-525, 2025-06-23:
### Have you reviewed the Frequently Asked Questions section available for your supported device in the wiki?

- [x] I have reviewed the FAQ for my supported device
- [x] I couldn't find the answer for this in the Frequently Asked Quetions section for my supported device

### If your issue is related to being able to run certain emulators, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

- [x] I have reviewed the Emulators and Ports section.
- [x] I couldn't find the answer for this in the Emulators and Ports section.

### Which device are you having this issue with?

R35S

### Description

How to compile and package Source code (tar. gz) into img；
For example, 20250531 Source code (tar.gz)

[image]

### Expected behavior

How to compile and package Source code (tar. gz) into img；
For example, 20250531 Source code (tar.gz)

### Steps to reproduce the bug

How to compile and package Source code (tar. gz) into img；
For example, 20250531 Source code (tar.gz)

### What version of ArkOS are you on?

05312025

christianhaitian, 2025-06-28:
Not possible.  To build this, you have to follow the instructions here (https://github.com/christianhaitian/arkos/wiki/Building) and even then, it's not a simple task.

closed 2025-06-28 by christianhaitian

3v1n0, 2025-11-23:
Are you aware that if you modified any source (including build scripts) that are GPL or LGPL code you must release those together with the binaries, right? And that once you mix all these (even linking together) into an unique work all becomes GPL and thus you've to release the full set of sources (including the scripts that makes them to be in an unique, single, blob that you eventually release), right?

RetroGFX, 2025-11-23:
> Are you aware that if you modified any source (including build scripts) that are GPL or LGPL code you must release those together with the binaries, right? And that once you mix all these (even [...]

Are u aware that 1, this isn't accurate, considering there's closed source programs, like drastic, that are included with permission from the developer, so are u saying that this automatically becomes open source?? Sorry, incorrect, so programs included follow their respective license. 
And 2, all the source is online in the repos.. the arkos build scripts are there, the patches are there,  the emulator and core scripts are.. the base is Ubuntu, so all that's there.. the license ark is under is mit, and it's packages all fall under their respective license. Emulationstation, portmaster etc, all have their licenses and that's what they follow.. there's no way to include 1 zip file that includes sources for everything in this os, or mine either.. I build from source, and I can tell you my source folder is about 50gb, so we're supposed to zip all those individual repos together and provide it in the image to users?? Not possible.. you want source, the repos are open and your free to clone and build